augisbud/m365_dashboard #1: Hello Kate - no Bus-

McKaNaN, 2018-08-21:
Hi.

I installed it and all done! :)

The problem, when I power on M365: LCD shows this message: "Hello Kate - no Bus -" 
Any idea?

McKaNaN, 2018-08-21:
https://youtu.be/q8CMcGD0eyE

augisbud, 2018-08-24:
Have you used the schematics provided?
https://camo.githubusercontent.com/a912641249173768ae60670e843c62294d06da4f/68747470733a2f2f656c656374726f2e636c75622f646174612f666f72756d2f6d657373616765732f34323633312f696d616765732f31313636302d313238302e6a7067

augisbud, 2018-08-24:
Gonna upload an updated version :)

augisbud, 2018-08-24:
Check out the new code provided, it's a revision from the original one, better ui and stability, however the arduino freezes once in a while, a watchdog seems to help it but not all the time, still looking for a solution 😕

McKaNaN, 2018-08-24:
Ok! I check it and comment here :)

McKaNaN, 2018-08-27:
I compiled new code and show this:

[image]

A friend say me that the connection is: 
[image]

Its correct?

augisbud, 2018-08-27:
[image]

augisbud, 2018-08-27:
I use this exact scheme without a problem

augisbud, 2018-08-27:
Send me a picture of the top part of your scooter with the pcb

McKaNaN, 2018-08-27:
Nothing. I cant connect BUS with screen.
All time shows same message: "BUS not connected! No data to display!"

I have question: 
[image]

This pins are correct?

augisbud, 2018-08-27:
Yes the pins are correct, can you unscrew the pcb and send me a pic

McKaNaN, 2018-08-27:
[image]
[image]
[image]
[image]

Connections are same at schematics :)

augisbud, 2018-08-27:
I meant the scooter's pcb

augisbud, 2018-08-27:
The blue one with the button

augisbud, 2018-08-27:
Ok, I'm going to flash the firmware to my scooter to make sure it's ok

McKaNaN, 2018-08-27:
[image]

I used this connection picture. I think that is the same that you but I cant understand where is the problem :(

augisbud, 2018-08-27:
Me neither, are those really 1n4148 diode, 120 resistor?

McKaNaN, 2018-08-27:
[image]
This is the resistor

Diode shows:
PH
41
48

Is correct?

augisbud, 2018-08-27:
You might wanna try a 1n4148 diode although they seem to be very similar

McKaNaN, 2018-08-27:
I dont know....
I need to go to buy new diode.
I change + and - of diode, and shows same message :(

OMG

chriskross, 2018-09-01:
Have you connect the same gnd for arduino and pcb?

McKaNaN, 2018-09-01:
Yes! The GND of Bluetooth PCB in GND of Arduino Mini Pro

closed 2018-09-14 by augisbud