augisbud/m365_dashboard #2: M365 languages translation

AntoineWaes, 2018-08-25:
Hey ! I can provide you my language.h file in French if you want to add it to the files ;) 
Maybe you could do a video showing how to plug and solder the wires 

Thanks for the sharing, and nice work dude !

AntoineWaes, 2018-08-25:
Thanks a lot for the screens ! The wiring doesn't seems to work for me, it keeps saying 'No bus connected" .. And the M365 shows the 4 LEDS on (with one green)...

augisbud, 2018-08-25:
Have you connected the resistors and the diode
?

AntoineWaes, 2018-08-25:
Yes, this is what I did, is it OK ? 
[image]

augisbud, 2018-08-25:
No let me get a piece of paper quick

augisbud, 2018-08-25:
https://photos.app.goo.gl/1Zbvogu9Ag9gwViG6

AntoineWaes, 2018-08-25:
This is mine : still not working, and this is what you sent .. What firmware do you have ?
[image]

augisbud, 2018-08-25:
1.4.0, do they have a common ground? I.E The ground and vcc of the arduino is soldered to the scooter

AntoineWaes, 2018-08-25:
Yeah they are both connected to the  Arduino from the scooter

augisbud, 2018-08-25:
Hmmm, ok, what chip is that?
https://i.imgur.com/VHPWohZ.png Here's how to connect it

AntoineWaes, 2018-08-25:
Do you know if there is a difference btw T4148 and 1N4148 diode ? Because I have a T4148

augisbud, 2018-08-25:
It might be the case, no, not really

augisbud, 2018-08-25:
I think that's something to do with the arduino not sure yet

AntoineWaes, 2018-08-25:
The resistor is a 120ohm right ?

augisbud, 2018-08-25:
Yes

AntoineWaes, 2018-08-25:
I try a different Arduino, now it is a Arduino Nano. Can you provide a picture of your installation ? With the Arduino ? It still doesn't work .. I'm pretty much desperated right now

augisbud, 2018-08-25:
Use the 1N4148 Diode, and the schematics provided in the wiki

augisbud, 2018-08-25:
I'm not home rn so I can't help

augisbud, 2018-08-26:
This is what I found browsing russian forums about your issue
---Russian Forums---
In general, I realized the problem with nano. CH340, which stands in them, does not fall into any sleep mode, but constantly pulls TX to plus through a 1kΩ resistor. Accordingly, the bus levels are shifted upwards and the scooter ceases to perceive them.

closed 2018-08-29 by augisbud