augisbud/m365_dashboard #3: Freezes once in a while

augisbud, 2018-08-25:
The Arduino freezes once in awhile it only seems to be happening with my hard driving so right now I think it's trying to open the menu because there is no speed check for that but that doesn't always seem to be the case. Anyone else experienced something similar?

augisbud, 2018-08-26:
I'm starting to think that the issue is with the library, gonna change it to the Adafruit library and maybe create a separate github for testing purposes 😕

augisbud, 2018-09-05:
Try out [this](https://github.com/augisbud/m365_dashboard/tree/beta)
This is the beta branch, it might contain bugs, but worth checking out, it uses a U8G2 library so download it before.

augisbud, 2018-09-10:
Well I have discovered that it seems like the bus overflows, because everytime i spin the tire in the air from 5km/h to 30km/h(custom firmware), the speedo freezes and sometimes catches back on

tibzejoker, 2018-09-10:
Oh, ok..
So maybe the solution may be to add an interruption timer that starts before every serial.read ? 
Then you can flush the serial port if it freezes by doing something like that if i remember : 
void flushSerial(){
  while(Serial.available () > 0){
     Serial.read(); 
  }
}

tibzejoker, 2018-09-10:
Or, maybe the best solution is to use 
Serial.onReceive(stockInATabFunction);
that triggers on every serial event ? 
then in the dataFSM(), you can just check if there is something waiting in the tab

augisbud, 2018-09-10:
Hmmm, It freezes because it receives too much data so a millis() delay could help it

augisbud, 2018-09-10:
But a delay is not a solution, so having something like you mentioned could work better

voodou59, 2018-10-07:
Did you tried with pro mini 3.3v ?

tibzejoker, 2018-10-08:
I have no idea, but can you give us the link to that video please ? 
i've also seen that some days, instead of freezing, it could work just nicely.. Btw, how have you found out that it is an overflow error ?
Thanks !

augisbud, 2018-10-10:
It's kinda an overflow, even the xiaomi mi app freezes if you put the scooter in the air and press the throttle fully

augisbud, 2018-10-10:
> Hi, thanks for sharing that project. Did you tried with pro mini 3.3v ?

You will destroy it, the rail is 5v, a capacitor or some kind of regulator.

voodou59, 2018-10-10:
Not really. It is supposed to accept voltage up to 12VDC.

voodou59, 2018-10-10:
I agree that 12v requires a heatsink but the electric scooter regulator range is 0 to 5.5-6v max

jordicli, 2018-10-11:
For me it's freezing in every trip :(

lukaszlukasz0, 2018-10-11:
For me it happend one time, i drive with this software over a week (60km).
Maybe it depends of arduino, ther are so many manufactors?

augisbud, 2018-10-11:
more like on the scooter I guess

tibzejoker, 2018-10-11:
Mine freezes every 1km approximatively..
I dont really know why because when i put the scooter in the air and press the throttle fully nothing more than the "normal" 1km freeze occurs..
I'm trying to add a second "watchdog arduino" to see if i can fix it that way

tibzejoker, 2018-10-11:
of course i have deleted the bootup par to skip the animation ^^.
I'll tell you if it can kind of fix it

augisbud, 2018-10-11:
Check if the percentage drops after sometime, because sometimes only the speedo freezes

tibzejoker, 2018-10-11:
no it doesnt unfortunately.. everything stops even the percentage.. (infinite power i gess ?? 👍 )

augisbud, 2018-10-11:
hmm, for me only the speedometer crashes

lukaszlukasz0, 2018-10-11:
and for me :-)

All scooter hang up, and stop driving, all led are off, and the display was on and suspended showed the speed.

tibzejoker, 2018-10-11:
hey guys i've figured out that when the port crashes, the arduino is still alive and working ! 
https://photos.app.goo.gl/jfD18F17mBE4PSTo9
On this video you can see that the screen is frozen and the arduino is still working (i've made the led on port 13 blink for debug)

lukaszlukasz0, 2018-10-11:
I see something like this sometimes (long time ago), when i drive with phone on bluetooth, my speed on mihome app freezes.
 I am not sure if this is not a coincidence, but often helped by tapping on a platform or passing through bumps or holes on the way :smiley:

tibzejoker, 2018-10-12:
I've finally fixed it with my second watchdog arduino.
Actually, when the arduino freezes due to a serial overflow, the 13th pin starts to blink pretty fast, and you cant reset it via the reset pin. You have to physically turn it off and on.
That's why i'm now doing that.
I'll release my fix tomorrow i think, it only costs the price of a second arduino nano.
[image]

tibzejoker, 2018-10-13:
It's not the best solution but  at least.. it works ^^
https://github.com/tibzejoker/m365_dashboard

carlosmellado, 2018-10-14:
Hi all,
I’m waiting for a new oled screen to test it out since I broke mine :p but...
There is a command for serial that checks for an overflow, and another for flush the serial, so... what if you put this piece of code at the very begging of the loop() function?

if(XIAOMI_PORT.overflow()) {
XIAOMI_PORT.flush();
}

Sorry if there are errors but I’m writing from the phone.
Will be working also to find a software based solution! :)

augisbud, 2018-10-14:
XIAOMI_PORT.overflow() doesn't really exist so we would need to define it.
XIAOMI_PORT.flush is the same as Serial.flush() so it could work.

augisbud, 2018-10-14:
https://www.arduino.cc/en/Reference/SoftwareSerialOverflow

toniht88, 2018-10-14:
> of course i have deleted the bootup part to skip the animation ^^.
how did you do it??

augisbud, 2018-10-14:
> > of course i have deleted the bootup part to skip the animation ^^.
> > how did you do it??

  display.setFont(m365);
  displayClear(0, true);
  display.setCursor(0, 0);
  display.print((char)0x20);

remove this.

vanekCZ, 2018-10-17:
Hello, just upload everything to arduin and everything will work? Or does something have to be fine-tuned? Can I translate the language somehow? I needed a Czech language. Thanks for advice.

carlosmellado, 2018-10-17:
> Hello, just upload everything to arduino and everything will work? Or does something have to be fine-tuned? Can I translate the language somehow? I needed a Czech language. Thanks for advice.

Please, keep this thread for the issue related.

vanekCZ, 2018-10-17:
> > Hello, just upload everything to arduino and everything will work? Or does something have to be fine-tuned? Can I translate the language somehow? I needed a Czech language. Thanks for advice.
> 
> Please, keep this thread for the issue related.

Better would be if you answered, then I will delete it.

jordicli, 2018-10-17:
> if(XIAOMI_PORT.overflow()) {
>XIAOMI_PORT.flush();
>}

It seems that overflow function it's defined in serial software library and not defined in serial hardware library so can't be used in our case :(

I found this topic [https://arduino.stackexchange.com/questions/19078/hardwareserial-check-for-overflow/19085](https://arduino.stackexchange.com/questions/19078/hardwareserial-check-for-overflow/19085) don't know if it can be useful

lukaszlukasz0, 2018-10-18:
I have idea, but i can't test it, because my display dont hang up ( i cant do fast that 30km/h on my engine)

Check 172 line of define.h
Speed can't be integer, because integer is to max 32,768 (32,7km/h).
It has to be unsigned int or long.

Test if that helps?

augisbud, 2018-10-18:
I will test it

augisbud, 2018-10-18:
Whenever it's unsigned int there are random speeds shwon

lukaszlukasz0, 2018-10-18:
Something like that happens with Arduino as we cross the variable barrier https://imgur.com/gallery/r9HgHd5

lukaszlukasz0, 2018-10-19:
I think that you could use our usb-to-ttl and see what happens on the line BUS? When this bug is happening.

Unfortunately, I do not have such possibilities because my scooter does not accelerate to this speed, but I will try to work with the Botox Software generator and set other parameters :wink:

jordicli, 2018-10-19:
In my case the freezing has nothing to do with going faster than 30 km/h, I'm on stock firmware 1.4.0 and can't go faster than 25 km/h

nebman, 2018-10-19:
Check your connections too. I've had the display freezing because of vibrations when testing with dupont connectors

lukaszlukasz0, 2018-10-24:
i make test code to calculate speed over 32.7, maybe this will help the screen not freezing.

Can anyone test it?
https://github.com/lukaszlukasz0/m365_dashboard

augisbud, 2018-10-29:
> i make test code to calculate speed over 32.7, maybe this will help the screen not freezing.
> 
> Can anyone test it?
> https://github.com/lukaszlukasz0/m365_dashboard

i need your opinion but i think it should look something like this

```
struct __attribute__((packed))A23CB0 {
  //32 bytes;
  unsigned char u1[10];
  int16_t            speed;              // /1000
  unsigned int  averageSpeed;       // /1000
  unsigned long mileageTotal;       // /1000
  unsigned int  mileageCurrent;     // /100
  unsigned int  elapsedPowerOnTime; //time from power on, in seconds
  int           mainframeTemp;      // /10
  unsigned char u2[8];
} S23CB0;
```
`
m365_info.sph = (double)((double)S23CB0.speed / 1000.0);  `

augisbud, 2018-10-29:
yes, i have read that yesterday, but i checked out other sources and everyone is using that

lukaszlukasz0, 2018-10-29:
maybe they use other processors that normally use int_32.

Speed https://imgur.com/gallery/57RGf1M

Check this pic, smartinick make display https://github.com/smartinick/esp32_xiaomi_m365

lukaszlukasz0, 2018-10-29:
m365_info.sph = (double)((double)S23CB0.speed / 1000.0);  

why do you assign the variable unsigned int to float (double)?

augisbud, 2018-10-29:
what's that telegram? can you send me a link

augisbud, 2018-10-29:
from more examination i can see that
_the controller spits out random speeds that cause the overflow, only happens when gas and brake both are applied_
_even if we store the speed properly, we won't solve the issue with random speed_
So @lukaszlukasz0 solution for calculating speed over 32.7 km/h would help, but we need to solve another issue.

augisbud, 2018-10-29:
to store speed like 39km/h and etc we need a 32bit integer like in java 

int32_t is the same as long, can store negatives numbers, range from -2,147,483,647 to 2,147,483,647
unsigned long is the same as uint32_t,can't store negative number, range from 0 to 2,147,483,647
```
struct __attribute__((packed))A23CB0 {
  //32 bytes;
  unsigned char u1[10];
  long               speed;              // /1000
  unsigned int  averageSpeed;       // /1000
  unsigned long mileageTotal;       // /1000
  unsigned int  mileageCurrent;     // /100
  unsigned int  elapsedPowerOnTime; //time from power on, in seconds
  int           mainframeTemp;      // /10
  unsigned char u2[8];
} S23CB0;
```
`unsigned long sph;`
`m365_info.sph = abs(S23CB0.speed) / 1000L);                 // speed`

lukaszlukasz0, 2018-10-29:
I think :-)

**long               speed;              // /1000** 
You can't do this, because speed from controller are in 16bit signed integer, not 32.

You need to take it in this format.

 unsigned long sph; 
 m365_info.sph = abs(S23CB0.speed) / 1000L);                 // speed  

Here is don't matter, because unsigned int is sufficient to store speed to 65km/h, becaause before we do abs() on him.

augisbud, 2018-10-29:
Ok, can you send me that telegram chat?

lukaszlukasz0, 2018-10-29:
view last post :-)

Do you have access to saved data from the bus?
Can you send it?

There is some such clean data here, but there is probably nothing about speed in them.
https://github.com/smartinick/esp32_xiaomi_m365

augisbud, 2018-10-29:
no i don't :(

pedrous1976, 2018-12-05:
good day, I have a plea. i have downloaded SW for display on m365 here from the site. The arduine version is 1.8.5.
When compiling, it writes an error.
C: \ Users \ kiki \ Desktop \ display \ M365 \ M365 \ M365.ino: In function 'void displayFSM ()':

C: \ Users \ kiki \ Desktop \ M365 \ M365 \ M365.ino: 421: 24: warning: switch condition has type bool [-Wswitch-bool]

        switch (WheelSize) {

                         ^
you can load into arduin, but in M365 the display will freeze and help reset.
Thanks in advance for your help ...

The project occupies 19526 bytes (63%) of the storage space for the program. Maximum is 30720 bytes.
Global variables occupy 715 bytes (34%) of dynamic memory, 1333 bytes remain for local variables. Maximum is 2048 bytes.

geirskogul, 2019-02-21:
> I've finally fixed it with my second watchdog arduino.
> Actually, when the arduino freezes due to a serial overflow, the 13th pin starts to blink pretty fast, and you cant reset it via the reset pin. [...]

I know this is an older reply, but I can't figure out what you're supposed to do with the nano for your fork, and you don't have issues enabled on your repository page :(.  I'm willing to deal with an older fork if it fixes the freezing issue.

Hermit12, 2019-02-22:
Same here, I tried to get the physical watchdog going. I added the missing defines so that the code seems to be complete, but when I connect it like on the diagram display doesnt work.
Can you please provide the fully working solution? Many thanks!

Vytciokaz, 2019-02-26:
Hi, I probably found why the screen freezes.
When I started the screen at the test stage, everything was fine, I drove about 30km and no one got stuck. When everything was assembled and the arduino put the case in the scooter, and I chose the thinner screen wires, the screen stuck immediately after the ride. When I disassembled - everything worked again. Choosing thicker wires to the screen - the screen just got stuck sometimes. 

I found such a thing that: The screen works with the I2C signal. When you extend the screen wires, thinner or longer. That is long for I2C. It means that the SDA and the SCL may cause noise and SDA and SCL. That's the least influence on each other. A single wire for the SDA and a single wire for SCL free, but that would work.
I have got some timeouts, but it is not full proof against I2C bus errors. I have seen a number of propositions in the library yet.
The I2C is also designed to be used in a pcb board.

You need either an arduino to install on the screen or look for another solution...
Here's an example of how I just freeze the screen with just a touch.
[](url) https://drive.google.com/open?id=1naDs08Jd8gVpxAuW47bFvRbOPc9G6R5D

geirskogul, 2019-03-03:
That would explain why it freezes a bunch on heavy regenerative braking - electrical noise.  I wonder if 

A: a ferrite bead or similar on the power lines would help, and

B: Putting the display right on the arduino with the shortest lines possible would help

augisbud, 2019-03-04:
I think B would be the answer, I'm thinking of attaching an led to see when it's the arduino overflowing, and when it's an i2c bus error

[38 more replies]