augisbud/m365_dashboard #21: Compilation Issue

nowisnowis, 2018-10-29:
Hi when trying to compile and uplode your code I get the following errors:

Arduino: 1.6.6 (Windows 10), Board: "Arduino Pro or Pro Mini, ATmega328 (5V, 16 MHz)"

C:\Users\svillalobos\Documents\Arduino\sketch_oct29a\M365.ino: In function 'void setup()':

M365:20: error: redefinition of 'void setup()'

 void setup() {

      ^

sketch_oct29a:10: error: 'void setup()' previously defined here

 void setup() {

      ^

C:\Users\svillalobos\Documents\Arduino\sketch_oct29a\M365.ino: In function 'void loop()':

M365:76: error: redefinition of 'void loop()'

 void loop() { //cycle time w\o data exchange ~8 us :)

      ^

sketch_oct29a:15: error: 'void loop()' previously defined here

 void loop() {

      ^

exit status 1
redefinition of 'void setup()'

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

Could you point me where can be the error

Thanks

Simon

augisbud, 2018-10-29:
Simply download the source, extract it to your desktop, open the M365 folder and then try to compile.

nowisnowis, 2018-10-29:
Thanks, that did it!, noob issue, just to let you know when you upload the libraries the sketch tab contains already a void setup and a void loop

closed 2018-10-29 by nowisnowis