augisbud/m365_dashboard #26: Rotate screen 180 degrees

serpro, 2018-11-10:
Hello

I have the dual colour screen "Yellow/Blue" and I noticed it would look much better if I had the yellow bar on the bottom rather than on the top. I can't see in the code any configuration for this.

Kind regards,
Sergio

carlosmellado, 2018-11-11:
This kind of screens doesn’t have any configuration for the color. The top LEDs are yellow and the other ones are blue, so, you can’t change them by software.

carlosmellado, 2018-11-11:
Or maybe you’re asking that in order to flip down manually your screen and then flip also the graphics on code 😅
On this case, you have to set this code on your setup, after initializing your display, before display.setFont(m365);

//Rotate the screen
display.setRotation(2);

I think the number 2 is the one you’re looking for, if not, try changing it!

Good luck!

LukCaha, 2018-11-11:
see this:
https://github.com/greiman/SSD1306Ascii/issues/23#issuecomment-396244537
and use this:
`display.displayRemap(true);`

serpro, 2018-11-11:
> see this:
> [greiman/SSD1306Ascii#23 (comment)](https://github.com/greiman/SSD1306Ascii/issues/23#issuecomment-396244537)
> and use this:
> `display.displayRemap(true);`

That worked! Thanks!

closed 2018-11-11 by serpro