augisbud/m365_dashboard #27: Configuration to use SPI OLED

myoung34, 2018-11-19:
Do you have any examples on how to use an SPI OLED that has 7 pins (mostly RST)

Hermit12, 2019-02-17:
Hi, I'm new to programming and like to use a [1.3'](https://www.aliexpress.com/item/1-3-inch-IIC-I2C-Serial-128x64-OLED-LCD-Display-White-Screen-Module-For-Arduino/32830144222.html?spm=a2g0s.9042311.0.0.5ec14c4djwYvEL) or [1.54'](https://www.aliexpress.com/item/1-54-inch-White-OLED-Display-Module-128x64-SPI-Interface-OLED-Screen-Board-3-3-5V/32837724627.html?spm=2114.search0104.3.1.4b134673jFPAJH&ws_ab_test=searchweb0_0,searchweb201602_6_10065_10068_10547_319_317_10548_10696_10084_453_10083_454_10618_10304_10307_10820_10821_537_10302_536_10902_10059_10884_10887_321_322_10103,searchweb201603_70,ppcSwitch_0&algo_expid=4d1234bb-d015-4664-b1a6-a95c27f7a033-0&algo_pvid=4d1234bb-d015-4664-b1a6-a95c27f7a033&transAbTest=ae803_4) Display instead of the 0.96' one.
The oleds I have in mind are hopefully compatible with SSD1306, I tested the 1.3' one and everything seems to work, but the whole content is set a little bit to the left. What do I have to change in your code to set it to the center?
Regards

augisbud, 2019-02-18:
@Hermit12  the resolution of your screen is bigger than 128x64, so you would need to adapt to that in the code, essentially shift everything or make everything bigger

Hermit12, 2019-02-18:
Thank you for your quick answer. I figured out that it will work if I modify the starting point 2 columns to the right, but I'm not quite sure how to do it. Resolution is the same but apparently all existing 128x64 OLEDs with SH1106 have the 128x64 OLED screen centered into the 132x64 display RAM, which leads to a two pixel shift, when using SSD1306 software. Have you any idea how to shift the starting point?

augisbud, 2019-02-19:
@Hermit12 you could try to change command like display.setCursor(0, 0); to display.setCursor(0+2, 0);

augisbud, 2019-02-19:
@myoung34 to use the SPI OLED, you would need to reconfigure the initialization, an example 
SPI initialization:
https://github.com/greiman/SSD1306Ascii/blob/master/examples/HelloWorldSpi/HelloWorldSpi.ino
I2C initialization:
https://github.com/greiman/SSD1306Ascii/blob/master/examples/HelloWorldWire/HelloWorldWire.ino

Using these examples you could try to do it, I don't have a SPI screen so I can't test it

augisbud, 2019-02-19:
@myoung34 to use the SPI screen, go to defines.h and change 
`//#define DISPLAY_SPI
#define DISPLAY_I2C`
to
`#define DISPLAY_SPI
//#define DISPLAY_I2C`

closed 2019-02-19 by augisbud

Hermit12, 2019-02-28:
Figured it out. If you want to use bigger screen with different chip (SH1106) you just have to edit display.begin:
`display.begin(&SH1106_128x64, 0x3C);` 
or for SPI `display.begin(&SH1106_128x64, PIN_CS, PIN_DC, PIN_RST);`
The library you use supports it OOB.
Maybe you can add this code as enhancement in your readme. I tested it with the 1.3'' I linked above.