augisbud/m365_dashboard #28: Bus not connected! No data to display!

nextor11, 2018-11-29:
Hi! I'm always getting "Bus not connected! No data to display!".

I've read through the closed issues and found one named "Hello Kate - no Bus-" where the problem seemed to be the resistor that was used, so I tried using different resistor values, first 120 ohm, then 150, then 100, but I always get that message. I also tried multiple 1N4148 diods, but no luck.

I know that the signal coming from the BLE module reaches the RX pad, so I don't know what might be the problem.

Any suggestions would be much appreciated. Thanks!

piyu23srma, 2018-12-07:
Hi,
I am using Arduino Nano for testing. All the connection are completed as shown in the schematic. But I am getting the error "BUS not connected! No Data to display!"

ronnymo15, 2019-01-06:
I also have the Problem, that the display shows "BUS not connected! No Data to display!". I tried it with 130 ohm, 100 ohm and 100+10+10 ohm resistors cause I had no 120 ohm at home - but nothing worked :(

afonsos, 2019-01-29:
I almost gone crazy with this,but solved with a resistor of 1% tolerance,that made it work

closed 2019-02-18 by augisbud

mikl912, 2019-03-15:
Hi! Ihave the same issue. 
I tried changing the resistor and diode but it didn't work.
I have an arduino copy: a wavegate pro mini.
Do you have an idea of what can cause the communication trouble?
Thanks