augisbud/m365_dashboard #32: Cruise Control Kers tail light

afonsos, 2019-01-29:
Im trying to change cruise control to ON,default is off.Change to ON then update cruise.
It stays ON but after next boot it goes to OFF again.
Im on an october release is this ok on recent updates? I didnt uploaded new code yet because Im not sure if its solved now and its a pain to get to arduino :)

augisbud, 2019-02-18:
It hasn't been patched yet, it will be

augisbud, 2019-02-19:
@afonsos I have patched it, give it a try

closed 2019-02-19 by augisbud

afonsos, 2019-02-19:
Thanks for the patch augisbud

Im getting this error compiling sketch

M365:235: error: conflicting declaration 'unsigned int c_speed'

   unsigned int c_speed; //current speed

                ^

M365:234: error: 'c_speed' has a previous declaration as 'long int c_speed'

   long c_speed; //current speed

        ^

exit status 1
conflicting declaration 'unsigned int c_speed'

augisbud, 2019-02-20:
give it a try @afonsos

augisbud, 2019-02-20:
for me it compiles, haven't uploaded it to the scooter

afonsos, 2019-04-10:
works as expected.
Thanks mate