augisbud/m365_dashboard #34: Restarts on turning light on

marioinz, 2019-02-19:
Hi. 
Mine dashboard keeps restarting (or turns of after a few restarts) when front light is turn on: https://youtu.be/BllbipDUYU8
My display is 4 wire SPI - this is only diffrence to your recomendations (i only changed defines and display initalization method). 
What may case such behavior ?

lukaszlukasz0, 2019-02-19:
Check the fuse f2 on mainboard.

augisbud, 2019-02-19:
@marioinz as Lukas said, check the second fuse on the main board, and try to flash the i2c code to see if it's the issue

marioinz, 2019-02-19:
ok. thx I will go with f2 fuse inspection ang get back heare with results. thank for quick response!

marioinz, 2019-02-19:
ok guys you were right. it is f2 fuse issue - i also have engine disruptions when lights is on. current resistance of fuse is ~2ohms.
what replace can i apply ? will this fit: http://electropark.pl/smd/3717-bezpiecznik-polimerowy-500ma-1812.html ? 
thanks in advance

augisbud, 2019-02-19:
@marioinz it's because the display is restarting, not sure what fuse you need

lukaszlukasz0, 2019-02-19:
@marioinz 500mA is ok. Like original..

marioinz, 2019-02-27:
thanks guys! fuse replaced and now everything works just perfect! i bought 3 sizes and 0805 fits perfectly. you can close issue. Best

closed 2019-03-04 by augisbud