augisbud/m365_dashboard #37: Power button issues

bouhady, 2019-03-03:
After connecting and assembling, the display is working fine BUT power button :
1. Not Toggling torch .
2. Not Turning of the "economic mode" (green led is on)
3. trying to powering off has no long beep - just when release the power button the scooter turned off.

geirskogul, 2019-03-03:
You've blown fuse F2

bouhady, 2019-03-04:
But when disconnecting the arduino the scooter works just fine?

bouhady, 2019-03-04:
> You've blown fuse F2

But when disconnecting the arduino the scooter works just fine?
The travel is ok and just when connecting the RX line the power button is not functioning.
If the Fuse was blown the problem was supposed to stay?

augisbud, 2019-03-04:
You may have shorted something

augisbud, 2019-03-04:
#34

bouhady, 2019-03-04:
OK. I'll try to replace this fuse and report.

geirskogul, 2019-03-11:
Also, make sure you're using a genuine nano or pro mini with the FTDI chips.  The MH348 chips don't pull up the signal properly and crash communications.

bouhady, 2019-03-11:
I used this Chip :
https://m.aliexpress.com/item/591964167.html
Do you think this was the problem?
(How should I pull up the RX well in that case?)

geirskogul, 2019-03-11:
Pulling up rx doesn't work unfortunately.

geirskogul, 2019-03-11:
That has the CH340, you need a genuine FTDI or an FTDI clone. When you
connect that one, does a red led start flashing on the dash? That's it
crashing

uvegp, 2019-04-14:
I've got the same issue (connected everything together, power button doesn't work, all 4 LEDs are glowing), but have not checked my fuses. I used an Arduino Nano clone from Aliexpress, had to remove the resistior from the original blueprint to get the data wire working. My workaround for the light issue was to add a switch to the circuit: first I power on the scoother, wait 5-6 seconds, then I power on the Arduino/OLED display circuit. This way I can turn the headlight on and I get the battery level displayed through the LEDs as well. Do you think I should still replace the F2 fuse? I mean if the setup is working this way, can I do damage to my scooter?

augisbud, 2019-05-07:
@uvegp you should probably still replace the fuse, for long term usage

closed 2019-05-07 by augisbud