augisbud/m365_dashboard #38: Possibility to add support for Ninebot ES1 / ES2?

NicholasTracy, 2019-03-04:
Hello,
 I have a ninebot ES1 that I have been playing with. I have tools available to reverse engineer the dashboard on the ES1 as well such as a scope with serial decoding and numerous Arduino Nanos and Minis.

Since both the M365 and the ES1 have the same connector to the dashboard and very similar design, does anyone know if its possible to use the same code to mod the ES1 or if it would be worth snooping the databus on the ES1 to see what is comming across? I'm actually very interested in taking it further and fully controlling the scooter with an arduino including throttle and brake commands and lighting as well as power and the alarm.

The dashboard is labeled G(black) T(yellow) P(green) 5(red) and I am assuming that G and 5 are ground and +5v. Not sure about the T and the P but I assume one is data bus and the other is the 36v power from the batter for the fuel gauge.   

Any input is appreciated before I attempt anything further.

augisbud, 2019-03-04:
From what I know, the ES1 / ES2 / ES4 have a different way of communication, just like the m365 pro

augisbud, 2019-03-04:
You could try, and what you will see from the start are the brake and throttle values, others you need to request for.

NicholasTracy, 2019-03-04:
Awesome, thanks. I will look at it at some point and see whats available.

closed 2019-04-30 by augisbud