augisbud/m365_dashboard #41: Hibernate feature

maniakpl, 2019-03-15:
Hi guys,
Wanted to add hibernate option to my dashboard. The dashboard must be unpluged from the scooter when changing or updating scooter's firmware, so hibernate option would be perfect for that.
Since I am new to programming, I need your help with this.
What lines of code should I put and where to get this feature?
Regards,
Bartek

augisbud, 2019-03-15:
i just have a toggle switch for the power of the arduino, not bothering with another menu option, although all you need to do is disable the packet request to accomplish this

closed 2019-04-04 by maniakpl

afonsos, 2019-04-10:
[image]

villamany, 2019-04-29:
This will do the job https://github.com/augisbud/m365_dashboard/pull/50