augisbud/m365_dashboard #43: How do I use uno instead of pro mini?

diqmwl, 2019-03-25:
How do I use uno instead of pro mini?

Do I have to change resistors or diodes?

Rosch94, 2019-03-25:
no im currently programming with arduino mega. use the same schematis. I would also recommand u to use the mega. With the mega u can open 2 serial communication one for the m365 and one for live debugging

NetBUG, 2019-03-25:
Uno is almost equivalent to Pro Mini, in worst case you will need to detach Serial adapter circuitry because of combined wire hack used.
Mega or Leonardo will be more convenient because of two ports but it's not necessary.
Anyway the live device should be as small as possible, thus full-size Arduino boards seem strange for it.

I would suggest to use Pro Micro as it's a minified clone of Leonardo.

diqmwl, 2019-03-26:
Mega rx tx What number did you use?

0 is an error :(

augisbud, 2019-03-26:
well, you need to disable the serial programmer on the uno, then you can use it, there is a lot of possibilities to do that

closed 2019-04-30 by augisbud