augisbud/m365_dashboard #46: Мигают светодиоды на плате скутера

lVlapuyc, 2019-04-16:
When the lights start flashing the LEDs on-Board a scooter. What could be the problem?

Sasha1804, 2019-04-16:
Надо заменить предохранитель на контроллере самоката (F2). Можно поставить нулевой резистор

lVlapuyc, 2019-04-16:
Я умею пользоваться паяльником, НО я полный ноль в этих контроллерах, схемах и т.п. Могу только по инструкции делать. Если не сложно, можете как-то более подробно сказать, где его заменить? F2 для меня дремучий лес)
И еще вопрос, я напечатал корпус из ABS, и попробовал его посадить на бесцветный сантехнический герметик. Так вот, к металлу самоката герметик отлично прилип, а к abs вообще не липнет. На что приклеивать?

lVlapuyc, 2019-04-16:
Что такое нулевой резистор? Можете ссылку дать, к примеру, в чип и дип (chipdip), чтобы я точно знал что купить?

Sasha1804, 2019-04-16:
https://www.chipdip.ru/product0/9000079834 такой резистор. Либо, еще лучше, самовосстанавливающийся предохранитель (любой), но размер должен быть 1206!!! 
Вот видео о замене: https://www.youtube.com/watch?v=ZleIXDNYiq4&feature=youtu.be

lVlapuyc, 2019-04-16:
Видео дает понять, как выпаять и как впаять. А где именно этот резистор находится на плате? Пжл, разжуйте мне более подробно, как для ламера, вы мой единственный русскоязычный помощник)

lVlapuyc, 2019-04-17:
Добрый вечер. Пара вопросов:
1. На видео используют предохранитель MF-NSMF050-2 BOURNS на 0,5А, номинальным напряжением 13,2В. Вы пишете, что подойдет любой, главное чтобы был 1206, непонятно, почему любой? Ведь другие могут быть рассчитаны на другие токи и напряжения, в таком случае теряется роль предохранителя.
2. На всякий случай поконкретней напишу, у меня мигают светодиоды на плате в руле и только тогда, когда свет включаю. Но, сегодня отпаял от этой платы назад все проводки от ардуино, и все мигание исчезло. Т.е. сам самокат и контроллер в порядке. Это точно проблема в предохранителе? 

Я могу предположить, что предохранитель видимо там стоит тоже самовосстанавливающийся и рассчитанный на определенный ток, и когда я подключаю монитор с ардуино, они на себя берут часть мощности, которая соразмерна или меньше мощности фары. Но когда я еще и фару включаю, с контроллера начинает браться больше мощности, сила тока возрастает и срабатывает предохранитель. Если это так, то тем более любой не подойдет. Но, я очень далек от всех этих электронных вещей, поэтому это только мое предположение.

augisbud, 2019-05-07:
since I don't understand any of this and google translate isn't much help either, I'm closing it, but for anyone else having the same problem, you have to change out the f2 (fuse). Check out #37 for english instructions/discussion.

closed 2019-05-07 by augisbud