augisbud/m365_dashboard #47: Is it possible for the Pro Mini to *send* commands?

flienky, 2019-04-17:
Well, if I understand correctly, the Pro Mini only reads the commands that the other microcontrollers send. 

Is it possible for the Pro Mini to send commands also, for example, to lock the scooter, or control the lights? Is there any problem with that approach?

Many thanks!

NetBUG, 2019-04-17:
As far as I understand, Mini Pro is a board with USB-Serial converter on board. Thus, the Tx pin of the ATMEGA chip sinks to converter and cannot provide enough power to drive the bus.
I would suggest to use other pins for bus interaction (e.g. pins 2-3 etc) and use SoftwareSerial instead of the Serial port.

Thus you change XIAOMI_PORT from Serial to SoftwareSerial in `defines.h` and use custom Rx and Tx pins when initializing it.

augisbud, 2019-04-18:
You can control the rear light, cruise control and the regenaration. :)
The arduino isn't stable enough for locking (it might freeze)

closed 2019-04-18 by augisbud