augisbud/m365_dashboard #48: Changing M365 firmware issue

villamany, 2019-04-28:
Any way for locking or hang the Arduino from the menu or at startup? I need to unpower the Arduino or disconnect the BUS wire for updating the firmware due the Arduino has the BUS bussy polling continuously data. I think a system for doing that would be very welcome without any need to add phisical switch or similar.

NetBUG, 2019-04-29:
I have done it in my branch: https://github.com/NetBUG/m365_dashboard/commit/f4529d95cd2296b94bd074f509d44b38e645f2fc
One thing is that I made a special PCB instead of tiny Arduino + dedicated screen, so the pin mapping for the display differs from the original one by @augisbud.

I have a button and an LED on the board, so the previous commit adds the code to handle button press.

villamany, 2019-04-29:
Nice, but what about to use a new menu item for that or maybe a booting test... IE: if arduino boots with the throttle pushed set the UART pins as inputs (hi impedance state) and goto sleep until next reboot?

villamany, 2019-04-29:
I have only basics of Arduino, but after first check the code, seems easy to implement. I will try to add that feature and post the result. If throttle > 150 at startup, dissable the UART and enter into infinite loop until next repowering.

villamany, 2019-04-29:
Done. Seems to works fine:
https://github.com/augisbud/m365_dashboard/pull/50

augisbud, 2019-04-30:
merged

closed 2019-04-30 by augisbud

afonsos, 2019-04-30:
Now that I have a switch?  :)
You guys are great,Thanks