augisbud/m365_dashboard #52: No data

rushtamas, 2019-04-30:
Hello,
I buld it and it says no data, and the screen flashing.
When I press the button on the arduino, rebooting continuously.
What could be the problem?

villamany, 2019-04-30:
Post a picture of your wiring and diode/resistor position

smidak, 2019-05-01:
I have the same problem. Pressing the adrduino reset button starts and does not function again when the power is cut off(no data)

NetBUG, 2019-05-01:
Hi guys,
this is 100% a hardware problem with the connection between the bus and controller.
Pressing any buttons won't work, since is does not change anything about the data.
If you use an ATMega32U8-based board - double-check the connection (diode and resistor) and ensure the value is correct (100-150 Ohm, not 150K! :) )
If you use a board with Mega328P - ensure that the pins are not puled down by the USB-TTL converter. If you have a board with that, it's advised to use SoftwareSerial on another pins (e.g. 2-3) rather than the hardware Serial.

villamany, 2019-05-01:
I'm using the diode and 220ohm resistor in TX line and another 220ohm resistor in RX line and working fine (i prefer to limit any current on both pins)

rushtamas, 2019-05-03:
I checked the diode polarity, the ohm resistance, tried with another board. Now "Bus not connected" screen shows, and can't turn off the M365 until I'm not plugging out the cables. But the screen not flickering, and not rebooting.

[image]

villamany, 2019-05-03:
Check the previous @NetBUG reply. 
You are using a NANO board with USB, the USB-UART chip uses the same TX and RX pins that you are trying to use for the M365 communication. The standard assembly uses an Arduino MINI tbat doesn't have USB.

rushtamas, 2019-05-03:
Okay, thx! I'll try it with arduino without usb on monday!

villamany, 2019-05-03:
As described by @NetBUG you can use software serial port on other pins or simply cut the traces fom TX/RX pins to the USB chip and use bootloader with an FTDI device or ICSP method

smidak, 2019-05-03:
[image]

smidak, 2019-05-03:
[image]

augisbud, 2019-05-07:
is the diode and the resistor correct? @smidak

smidak, 2019-05-08:
Yes, 2x220ohm+ TX diode😭

closed 2019-07-03 by augisbud