augisbud/m365_dashboard #66: Russian gibberish

tokfrans03, 2019-07-12:
Hello, ive been trying to get this working for a while now but it's not going too well.

I'm getting this on my I2C screen after starting [https://i.imgur.com/nW5RPlj.jpg](Image). And according to my russian friend its nothing but gibberish

I've imported all the LIB's and put em in the documents folder and followed all other instructions.

What can I do?

LukCaha, 2019-07-12:
Connect it with M365 ;-)

[image]

tokfrans03, 2019-07-13:
That shouldn't be necessary tho right?

NetBUG, 2019-07-13:
Hi guys!
I know the Russian localization looks weird. However, it works fine on the screen -- there is a feature of the fonts used. They do not follow neither UTF-8 convention (what a surprise for a bitmap font!), neither any of 8-bit encodings (ASCII, KOI etc). Unfortunately, the conversion is too complicated to be done on the go at compile-time through inlines -- or the macro would be huge.

If it's necessary, I can make a short code that converts from Unicode at runtime for these fonts, it will take about ~140 bytes of RAM. However, I'm surprised that the question was not raised by any Russian-speaking developer.
P.S. I do not know where to get additional fonts in case there is a request for Arabic/Hebrew/Norsk/Korean or any other languages support.

tokfrans03, 2019-07-15:
I just want it to work. I'm willing to help you out in any way possible

NetBUG, 2019-07-15:
Sorry, I didn't get your issue — I thought you're talking about unreadable strings in the code.

I have to check but the snapshot from early April must work correctly (language RU in language.h + System5x7mod.h as a font source — I tested it but reverted to English as it reads easier). The updates might have changed something.
I'll be by my scooter on Friday, so I can't check earlier, sorry

tokfrans03, 2019-07-15:
Oooo I think I fixed it! I just used the default "Adafruit5x7" font mentioned in some example code for the screen instead of "defaultfont". [Image](https://i.imgur.com/5XdflrQ.jpg)

closed 2019-07-15 by tokfrans03

thehair, 2019-08-09:
Just fyi for anyone else having this issue. I believe i was accidentally using the russian font and for some reason i couldn't get it to use the english font. I changed the following line in the defines.h from "#if Language == RU" to "#ifdef LANG_RU" and that fixed my issue