augisbud/m365_dashboard #70: Getting a green led with blinking tail lights

duffann, 2019-11-11:
Hello,

I have a strange problem. I power the arduino uno directly from the battery with a dc-dc converter with 12v, common the grounds , and using 120r and 1n4148 to connect to arduinos rx and tx with databus. When I power on the scooter while the ardunio is already on, I get a green led indicator with blinking taillight. ( I assume the connection between the esc and ble is borken) , as soon as I remove the connection from rx and tx, everything is ok with the scooter. 

I need to power the arduino from the main battery not from the ble.
What is wrong with my setup ?

Thanks

MathieuBre, 2019-11-13:
Hi,
The trouble you are experiencing is due to the voltage level on the arduino Uno on RX/TX pins. Indeed, the arduino Uno is forcing an average voltage higher than the bus which avoid the normal operation of the scooter (all data appear to be 0 if you check from the application). On an Arduino nano, this issue doesn't exist.

duffann, 2019-11-13:
> Hi,
> The trouble you are experiencing is due to the voltage level on the arduino Uno on RX/TX pins. Indeed, the arduino Uno is forcing an average voltage higher than the bus which avoid the normal [...]

What If I use only the microcontroller : Atmega328p isntead of the ardunio ?

MathieuBre, 2019-11-13:
the ATmega328p is the microcontroller of the Arduino nano, hence it should work properly. The "p" in this version stand for low power. Also, don't forget to add the diode/resistor on TX pin, and one resistor on RX pin (avoid to drain to much current).

duffann, 2019-11-13:
> the ATmega328p is the microcontroller of the Arduino nano, hence it should work properly. The "p" in this version stand for low power. Also, don't forget to add the diode/resistor on TX pin, and one resistor on RX pin (avoid to drain to much current).

Thanks for the answer, It is also the microcontroller of Ardunio uno but I dont think there will be a problem as you say. I am using the connection schematic in this github It only shows one resistor and diode on tx pin and nothing in rx, which resistor should I use for rx ?

MathieuBre, 2019-11-13:
Checking on the solved issues (https://github.com/augisbud/m365_dashboard/issues/52), it seems that a 220Ohm resistor should do it. I had the same issue as you using an Arduino Uno, switching to a Nano solved the communication trouble (at least the green led issue you are having).

duffann, 2019-11-13:
> Checking on the solved issues (#52), it seems that a 220Ohm resistor should do it. I had the same issue as you using an Arduino Uno, switching to a Nano solved the communication trouble (at least the green led issue you are having).

So I will try removing arduino uno and use its microcontroller only atmega328p and use 220ohm + diode on tx and 220ohm on rx.

duffann, 2019-11-13:
> Checking on the solved issues (#52), it seems that a 220Ohm resistor should do it. I had the same issue as you using an Arduino Uno, switching to a Nano solved the communication trouble (at least the green led issue you are having).

by the way , are you using 120 or 220 on nano ?

MathieuBre, 2019-11-13:
i used 120Ohm. if this is still not working for you, i suggest you use a SerialSoftware instead of the classical RX/TX pins.

duffann, 2019-11-13:
> i used 120Ohm. if this is still not working for you, i suggest you use a SerialSoftware instead of the classical RX/TX pins.

Thanks for all the answers it was a big help. Last question, if we switch to software serial, anything needs to change on the code side ? I couldnt find where we declared them.

MathieuBre, 2019-11-13:
You need to replace the line "#define XIAOMI_PORT Serial" from the defines.h file by:
`#include <SoftwareSerial.h>
SoftwareSerial XIAOMI_PORT(2, 3); // RX, TX`

Also it will free the classical serial for debugging.

duffann, 2019-11-13:
> You need to replace the line "#define XIAOMI_PORT Serial" from the defines.h file by:
> `#include <SoftwareSerial.h> SoftwareSerial XIAOMI_PORT(2, 3); // RX, TX`
> 
> Also it will free the classical serial for debugging.

Perfect! Thanks very much. Will share the result tomorrow.

closed 2020-06-16 by augisbud

MaeseppTarvo, 2021-05-03:
@duffann I know it is an old thread but can you tell me did you manage to get it to work? It seems that all the data is moving but the software do not know what to do with it.
I tried to change the #define XIAOMI_PORT Serial with #include <SoftwareSerial.h> SoftwareSerial XIAOMI_PORT(2, 3); // RX, TX but it gives me an erros:

```
In file included from /Users/xistence/Desktop/m365_display-master/nano_just_speedometer/nano_just_speedometer.ino:1:0:
sketch/defines.h:15:29: warning: extra tokens at end of #include directive
 #include <SoftwareSerial.h> SoftwareSerial XIAOMI_PORT(2, 3); // RX, TX
                             ^~~~~~~~~~~~~~
/Users/xistence/Desktop/m365_display-master/nano_just_speedometer/nano_just_speedometer.ino: In function 'void setup()':
nano_just_speedometer:6:3: error: 'XIAOMI_PORT' was not declared in this scope
   XIAOMI_PORT.begin(115200);
   ^~~~~~~~~~~
/Users/xistence/Desktop/m365_display-master/nano_just_speedometer/nano_just_speedometer.ino: In function 'void dataFSM()':
nano_just_speedometer:205:13: error: 'XIAOMI_PORT' was not declared in this scope
       while(XIAOMI_PORT.available() >= 2){
             ^~~~~~~~~~~
nano_just_speedometer:236:13: error: 'XIAOMI_PORT' was not declared in this scope
       while(XIAOMI_PORT.available()){               //read available bytes from port-buffer
             ^~~~~~~~~~~
/Users/xistence/Desktop/m365_display-master/nano_just_speedometer/nano_just_speedometer.ino: In function 'void sendQueryFromTable(unsigned char)':
nano_just_speedometer:538:3: error: 'XIAOMI_PORT' was not declared in this scope
   XIAOMI_PORT.write((unsigned char*)&buf, DataLen + 2);     //DataLen + length of cs
   ^~~~~~~~~~~
exit status 1
'XIAOMI_PORT' was not declared in this scope
```