augisbud/m365_dashboard #75: Greek Symbol Font

rediculum, 2020-06-02:
Hi

After successfully flashing the code, I get some kind of greek symbols as font. In language.h I have
```
#define LANG_ENG //English
``` 
and in defnines.h
```
#define DISPLAY_I2C

#define US_Version
```
[image]

nextor11, 2020-06-02:
That greek is definitely russian.

It's telling you "bus not connected". Your arduino is not communicating with the scooter.

rediculum, 2020-06-02:
@nextor11 I knew the message because I haven't connected it yet to the scooter, but now do I change the language to english?

rediculum, 2020-06-03:
@nextor11 thanks, that changed to english

closed 2020-06-03 by rediculum