augisbud/m365_dashboard #82: Can not get data (tried on Micro and Leonard)

MaeseppTarvo, 2021-05-03:
Hey guys! Can someone lead me somewhere...

I am using exatcly the same components with same connection as mentioned everywhere: 1N4148 Diode, 0.25w 120ohm Resistor.

For me It seems that all the data is moving but the software do not know what to do with it(Arduino).
I tried to change the
`#define XIAOMI_PORT Serial` with/to `#include <SoftwareSerial.h> SoftwareSerial XIAOMI_PORT(2, 3);` but it gives me an erros:

`'XIAOMI_PORT' was not declared in this scope`

With multimeter I tested against ground that data exists and the TX output value was about 3.001v

augisbud, 2021-05-03:
Are you using the original M365?
What's your DRV version?

MaeseppTarvo, 2021-05-03:
@augisbud Yes, I am using original M365 the EU version with custom firmware. DRV is 155, BMS 126 and BLE 090. Thank you for your very quick response 🙂

augisbud, 2021-05-03:
I'm not sure where you should start, maybe try their code with the lcd, just change the lcd printing to Serial. I can't really guarantee it will work.
I have acquired the M365 Pro, I believe I still have all the hardware and I will try to hook everything up to test since I see this is still being used by people.

MaeseppTarvo, 2021-05-04:
@augisbud It is good to know that you have the ability to test it, it would be awesome to know if you as a project author would be able to get correct values.

Also wanted to mention that to try with minimal code I took [this](https://github.com/fogbox/m365_display/tree/master/nano_just_speedometer) repo but everything is actually same with yours. But changing `#define XIAOMI_PORT` Serial to `#define XIAOMI_PORT Serial1` gave me some numbers and communication started :) Also had to change `UCSR0B &= ~_BV(RXEN0);` to `UCSR0B &= ~_BV(RXEN0);` because otherwise you get an undefined/undeclared errors. Unfortunately I have no idea what difference it makes or why those changes needs to be done?

Dubstaaa, 2021-11-15:
Im sorry for my low level English... I made possible to read all values on pro 2, 1s, essential and 3 including speed value. everything works fine now! 
visit my GitHub page, here's soon I will release the version v0.B

Dubstaaa, 2021-11-25:
Im glad to share with you my project, fully compatible with multiple scooter versions.
Neat dash, test release by Bruno. Available on https://github.com/Dubstaaa/NeatDash-